evilC/AutoHotInterception #13: [Help Wanted] Problem with key remapping

leobel96, 2018-04-26:
When I try to remap one key to another (for example X::A) it works without problems. If I try to remap a key to something different (a MsgBox, a Tooltip,...) it doesn't works. For example, in this code:
```
#NoEnv  ; Recommended for performance and compatibility with future AutoHotkey releases.
#SingleInstance force
#Persistent
#include <AutoHotInterception>

AHI := new AutoHotInterception()

keyboardId := AHI.GetKeyboardId(0x1B4F, 0x9203)
cm1 := AHI.CreateContextManager(keyboardId)

#if cm1.IsActive	; Start the #if block
X::
MsgBox, Hi
#if
```
A message box should appear when i press "X" but nothing happens. If I try using Subscription:
```
#SingleInstance force
#Persistent
#include <AutoHotInterception>

AHI := new AutoHotInterception()

keyboardId := AHI.GetKeyboardId(0x1B4F, 0x9203)
AHI.SubscribeKey(keyboardId, GetKeySC("X"), true, Func("KeyEvent"))
return

KeyEvent(state){
	MsgBox, Hi
}

^Esc::
ExitApp
```
Everything works as expected. I don't know what could be the problem...

evilC, 2018-07-02:
Sorry, dunno how I missed this.
No idea why that would be the case - your example works fine on my own PC when I substitute my own VID and PID

evilC, 2019-01-30:
I am going to close this now, due to inactivity. Pls re-open if you still need help

closed 2019-01-30 by evilC