evilC/AutoHotInterception #20: Add support for Unsubscribing

evilC, 2018-07-09:
Allow subscriptions to be removed in Subscription Mode

evilC, 2019-01-07:
I don't understand what you are saying. Are you maybe thinking that subscriptions persist between runs of scripts? They don't.
This is basically the equivalent of an AHK command like `Hotkey, F1, Off`, which is not a very common thing to do.

MartinLichtblau, 2019-01-08:
If so, then I have to change the organization of my scripts. I wanted to keep one global AHI object for all scripts for efficiency reasons. 
One feature is to scroll by moving the mouse, as long a certain button is pressed. So if it is pressed do `AHI.SubscribeMouseMoveRelative(trackpadId, false, Func("MouseEvent"))`. But what to do if the button is released? I had it all in one script with many other hotkeys and I run it only once on win-login. So, should I put the AHI-related code in another script and exit in on release?

commit 7c5c52c7 (2019-02-10): Add Unsubscribe endpoints. Fixes #20
closed 2019-02-10 by evilC