evilC/AutoHotInterception #24: Sending special characters via callback

Revvilo, 2018-12-05:
Some of the macros I make are to create shortcuts to special characters such as ✔. (Don't know the technical term for them)
But currently those characters don't seem to send correctly from within a callback. For example that tick comes out as `âœ”` when using any of the sending options available in AHK.

I don't really know enough about character formatting and inputs to guess why it'd be happening, but here's a script I made to reproduce it.
```
#SingleInstance, Force
#Persistent
#Include Lib\AutoHotInterception.ahk

AHI := new AutoHotInterception()

; Use your keyboard's handle
AHI.SubscribeKey(AHI.GetKeyboardIdFromHandle("HID\VID_03F0&PID_0024&REV_0130"), GetKeySC("F1"), true, Func("CallbackFunction"))

CallbackFunction(state) {
    If !state
        Return

    SendInput, ✔
}
```
I'd be interested in any workaround if you know of one.

evilC, 2018-12-05:
Does `F1::SendInput, ✔` suffer from the same problem? If so, it's nothing to do with AHI

Revvilo, 2018-12-05:
No, it doesn't, made sure to check that first. Sorry I was a bit tired when writing that and must've spaced out on it.

Revvilo, 2018-12-05:
Ah okay, I'd guess that this might be the normal behaviour of the character and AHK just has a system in place to allow for what I did in `SendInput, ✔` under normal circumstances, which these, I guess, are not.
I replaced the tick with the Unicode `{U+2713}` for it and it works perfectly.

Hopefully this will serve as some help to anyone who has the same problem in the future. Perhaps mention it in the docs?

closed 2018-12-05 by Revvilo