evilC/AutoHotInterception #26: External Mice stopped working

MartinLichtblau, 2019-01-10:
I had an external USB and Bluetooth mouse and both stopped working. Since this happened within the timeframe I assume it's because of AHI, or rather the Inception driver. 
I tried to fix it with the usual approach: on/off, reboot, unplug, uninstall. No luck so far. 
When I turn the mice on/off they popup in Device Manager as HID-compliant mouse, but don't work, and are not shown in Monitor.ahk.

evilC, 2019-01-30:
Define "Not working". Do you mean that they no longer function in Windows as mice, or do you mean that AHI does not see them?  
This seems unlikely to be related to AHI, or even Interception.  
What happens if you uninstall Interception?

MartinLichtblau, 2019-02-01:
They didn't respond. Didn't show up in AHI, through e.g. your Monitor.ahk. But were shown in Windows Device Manager. But more I didn't check. And now I can't reproduce, i.e. they work again. 
You can close the issue and I will reply in case it happens again. 
P.S.: Thanks for this wonderful software. It's really of great use for me! :-)

MartinLichtblau, 2019-02-01:
Indeed the mouse stops working after some hibernations. BUT the same happens when turning the mouse on/off multiple (~7) times. Once this single mouse stops works, other mice aren't detected as well. I used a bluetooth mouse and after it stopped working I plugged in a USB mouse. After a clean restart everything works normal again.

MartinLichtblau, 2020-03-26:
@evilC Is it somehow possible to restart the interception driver so it resets the counter without having to do a restart? 
Or more broadly, is there any way to keep using an external input device which you plugin/out often on a PC that often goes into standby; ergo a typical notebook scenario?

aldermath, 2020-11-04:
I have the same issue. After disconnecting and reconnecting an HID-compliant mouse of any kind a few times, inputs stop working even though the device shows up in device manager. AHI seems to interfere with it in some way even when no AHI script is running. I uninstalled the AHI driver and the issue no longer occurs. Not sure why this happens.

evilC, 2020-11-05:
> @evilC Is it somehow possible to restart the interception driver so it resets the counter without having to do a restart?
No

> AHI seems to interfere with it in some way even when no AHI script is running
This is because the Interception driver is always loaded, even when no AHI script is running. There's nothing I can do about it - it's a "feature" of Interception, and interception is closed source, so I cannot fix it.

xmaxrayx, 2024-05-25:
https://github.com/oblitum/Interception/issues/145

xmaxrayx, 2024-05-25:
anyone know a good alternative Interception driver ?

paulpv, 2024-10-15:
> anyone know a good alternative Interception driver ?

No. Would love to know! 🫤