evilC/AutoHotInterception #28: Investigate issues with devices stopping working after hibernate

evilC, 2019-01-30:
It seems maybe devices get a new ID after the PC comes out of sleep / hibernate?  
https://www.autohotkey.com/boards/viewtopic.php?p=260917&sid=fd0c2ccd278326e967e6329896bb7ad8#p260917

MartinLichtblau, 2019-08-09:
So I've faced this issue many times now and when I did I always did a restart, since that's the only thing that works; Replugging, Disabling and Uninstalling don't work.

Any external mouse stops to work when I unplug it and replug it, not when hibernating; so this changed. When this happens the device also isn't shown anymore in the Monitor.

evilC, 2019-08-09:
This is a known issue. Unplugging and replugging a device will cause the ID to increment.  
Once it increments to beyond ID 10 (For keyboards) or 20 (For mice), it will cease to function  
There is nothing I can do to fix this, it is an issue with Interception, and he will not fix it.

MartinLichtblau, 2019-08-10:
Ohh,... 
Could we restart the Inception driver? Perhaps this would reset the ID count, reinitialize it or so.

getSicko, 2021-05-01:
for some reason just restarting doesnt fix it for me. I have to unninstall driver first and then restart.