evilC/AutoHotInterception #33: Stuck Keys

MartinLichtblau, 2019-04-01:
I have remapped all modifier keys with AHI, for instance Alt is Ctrl now. But too often I get stuck keys, pressumably when pressing multiple modifier keys at the same time. 
Could it be that AHI's ability to block inputs is causing this? E.g. when a key is physically released while the input is blocked.

MartinLichtblau, 2019-05-14:
Finally I found the cause of it and a hack to fix it.

MartinLichtblau, 2019-05-14:
@evilC Am I doing something wrong. Or is there a simpler way to fix this? Or have you even already fixed it in a newer release?

evilC, 2019-05-14:
Yes, the thread pooling is a matter of concern for me too  
Crumbl3d submitted a PR with some code that makes all the callbacks fire in sequence - he currently added it as an option IIRC, but I was planning on making it the default.  
It looks like it is yet to be merged...  
https://github.com/evilC/AutoHotInterception/pull/32/commits/5dc35cc8d8feec456fa1da983692fe62bfab573c#diff-a4ebf4dc9a0414115220de3f4d687070L122  
If you can compile yourself, fine - if not I will try and make a build for you tonight to see if it solves your issues.  
Thanks for doing the legwork on this!

evilC, 2019-05-14:
I made a test release: https://github.com/evilC/AutoHotInterception/releases/tag/v0.4.0  
Notice the extra `concurrent` param for the subscribe methods. Try setting this to true and see if any issues still repro.

MartinLichtblau, 2019-05-15:
Thank you so much! 
The good news is that I can't reproduce the bug, neither with the concurrent param true, false or none at all. I'll try out some more in the next days, but this one looks goog now either way; although I don't know why.
Furthemore the event state value seems to be inverted; now 1 is up and 0 is down.

evilC, 2019-05-15:
> The good news is that I can't reproduce the bug, neither with the concurrent param true, false or none at all  
Hmm, strange. There were quite a lot of other changes and fixes in there - I did actually put some fixes in there for extended chars, it was maybe that...  
State inversion has been fixed.  
0.4.1 has been released

MartinLichtblau, 2019-05-17:
0.4.1 works fine! Interestingly the first thing I noticed with both new versions is the responsiveness. Typing feels so much smoother now. Before it also had a noticable delay. Don't know why and also don't need to know, but this is a huge improvement for me. 

#bug: sending RShift instantly kills my script.

evilC, 2019-05-17:
> #bug: sending RShift instantly kills my script.
If this is persisting, can you raise another issue please?