evilC/AutoHotInterception #37: If multiple mouse buttons change state in one update, only one event will fire

evilC, 2019-06-03:
Mouse button state is encoded in bitwise flags, and if really close together, **multiple**, button events can happen in one update  

LMB press = 1
LMB release = 2
RMB press = 4
RMB release = 8

LMB released **and** RMB released = 10 - **in this instance, only RMB release fires**

evilC, 2019-06-03:
@crumbl3d FYI I just found a major issue with the button processing - it's possible for more than one button or wheel to change state in one update, and the old code was only processing one thing  
I have [implemented a fix in UCR](https://github.com/evilC/IOWrapper/tree/hotfix/interception-mouse-button-multiple-update), if it all checks out, I will port to AHI  
So if you are still using your fork, you will probably want this fix ;)

crumbl3d, 2019-06-04:
@evilC Thanks for the heads-up! I think I merged the changes from this repo into my fork after the pull request was merged. Regardless, I am probably going to write my own C++ app for my needs as AHK is not flexible enough (maintenance-wise) for adding new features and responding to certain events. I was having problems with VMware mouse grab and interception making a mess fixable only by restarting or signing off. Basically sometimes only one window would react to my mouse input and everything else would be blocked. I have no clue why that was happening but I realized it was happening with VMware open the most.
TLDR: Thanks, will keep an eye on things! If you need any help I will try to respond when I have time. I am busy with exams and whatnot...

commit b512b6d5 (2019-06-08): Fix #37
closed 2019-06-08 by evilC