evilC/AutoHotInterception #48: Disable cursor and read raw input from mouse, how to do it ?

machomuchacho, 2019-10-11:
Hi, sorry if I'm being ignorant but I'm only learning all these things. I'm trying to use AutoHotInterception and UCR to fully emulate the DualShock 4 controller in order to play games on the PS Now, which only supports official PS4 controller. My only problem is that when I use DeltaToAxis plugin with mouse as an input and DS4 right stick as output, the mouse is still visible and moves around the screen even though the controls work. It shows up the PS Now overlay, and simply hiding the mouse or trapping it in the middle of the screen doesn't solve the problem. So I was trying to disable the cursor completely and do something to catch the low-level mouse delta only. What must I do ?

Thanks,

evilC, 2019-10-11:
There's no point in using AHI and UCR, as UCR already has it's own Interception implementation.  
AHI is for using Interception from AutoHotkey, which is a completely different scenario

machomuchacho, 2019-10-11:
Thank you, I guess I was getting it wrong^^

closed 2019-10-11 by machomuchacho