evilC/AutoHotInterception #50: Dear EvilC,Can you make a vedio tutorial about AHI?It's a huge PITA to set up! lol

MikuGenius, 2019-10-13:


evilC, 2019-10-13:
What do you mean by "Set up"? All you do is install interception, download AHI, copy one DLL into a folder, then run the example script

goodentree, 2019-10-14:
The installing interception on windows 10 is the difficult part.   Despite running command prompt as an admin, it will error that is can't write to c:\windows\system32.    Taking ownership of the windows\system32 folder isn't an option.  if I remember, I think I used safe mode.
Is it a dependency/requirement for writing to system32 vs running it from a different location? 
The manual CLI setup after requires a long attention span to setup individual....

What was I talking about again?

MikuGenius, 2019-10-14:
> The installing interception on windows 10 is the difficult part. Despite running command prompt as an admin, it will error that is can't write to c:\windows\system32. Taking ownership of the [...]

I hope that who can make a vedio tutorial,it's really difficult to understand.

evilC, 2019-10-15:
I seem to remember having seen that issue before, can't remember what the root cause was tho.
It's probably best to ask around in the AHI Discord channel, someone will probably know

TaranGenius, 2019-11-03:
> What do you mean by "Set up"? All you do is install interception, download AHI, copy one DLL into a folder, then run the example script

Thanks,I will have a try.

TaranGenius, 2019-12-09:
> What do you mean by "Set up"? All you do is install interception, download AHI, copy one DLL into a folder, then run the example script

Thank you so much,you are my savior!I finally work out.AHI is the best solution,it's awesome!
(Taran's method have a big bug which drive me crazy)

closed 2019-12-09 by TaranGenius