evilC/AutoHotInterception #51: UnsubscribeMouseMove disables mouse input...

manciuszz, 2019-11-25:
I've encountered an issue where "UnsubscribeMouseMove" disables mouse input when I suppose its intended behaviour is just to "unsubscribe" the callback?
```ahk
class Example {
        static device := {}
        __New(keyboardHandle, mouseHandle) {		
		this.AHI := new AutoHotInterception()

		this.device.keyboardId := this.AHI.GetKeyboardIdFromHandle(keyboardHandle)
		this.device.mouseId := this.AHI.GetMouseIdFromHandle(mouseHandle)
	}	
        blockMouseInput(block := false) {		
		this.AHI.SubscribeMouseMove(this.device.mouseId, block, ObjBindMethod(this, "__eventCallback", block))
		return this
	}
	
	unsubMouseInput() {		
		this.AHI.UnsubscribeMouseMove(this.device.mouseId)
		return this
	}
	
	__eventCallback(block, x, y) { 
		; ... empty fn
	}
        ...
}
Example.blockMouseInput(true) ; this would subscribe to mouse movement and block at the same time -> works as intended
Example.blockMouseInput(false) ; this would subscribe to mouse movement, but NOT BLOCK the mouse input -> works as intended

Example.unsubMouseInput() ; THIS IS SUPPOSED TO "Unsubscribe" from mouse movement and remove any block associated with it, YET it disables mouse movements completely (i.e locks out your mouse) and Example.blockMouseInput(false) or Example.blockMouseInput(true) wouldn't bring it back until you exit/reload the script...
```

I thought it was an intended behaviour at first, but after checking "SubscribeKey" to see if it acts similarly - it doesn't - after using "UnsubscribeKey" I can still use that key normally.

evilC, 2019-11-25:
Thanks for the heads up, I will look into it  
I can think of plenty ways that it could happen
BTW, `static device := {}` looks a little risky to me, if you have two instances of the same class, they will use the same static copy

manciuszz, 2019-11-25:
> BTW, `static device := {}` looks a little risky to me, if you have two instances of the same class, they will use the same static copy
It's just an example piece of code - in my real code that class is a singleton and that object holds only mouseId and keyboardId - nothing else, but thanks for the heads up anyways :)