evilC/AutoHotInterception #52: Hello EvilC,How do I edit Context Example.ahk when I add 3rd keyboard?

TaranGenius, 2019-12-09:
I try to change id1 to id2 and cm1 to cm2,but my 3rd keyboard still can't work.
Please help me out here,thank you so much.
[image]

evilC, 2019-12-09:
That script only has one keyboard in it, so is no different to the example?

PotatoJet, 2019-12-10:
> That script only has one keyboard in it, so is no different to the example?
I'm sorry that I didn't get the full screenshot，here it is.My 3rd keyboard just can't work with function.
How to edit 3rd_keyboard script?I really appreciate.
[image]

PotatoJet, 2019-12-10:
If you need,I even can pay the money,because I really need 3rd_keyboard fulled with functions.

PotatoJet, 2019-12-11:
> That has unreachable code
> 
> ```
> cm1 := AHI.CreateContextManager(id1)
> return ; EXECUTION STOPS HERE
> 
> ; ...
> id2 := AHI.GetKeyboardId(0x25A, 0x001F, 1) ; THIS LINE NEVER EXECUTES
> cm2 := AHI.CreateContextManager(id2)
> ```
> 
> Instead you want like
> 
> ```
> #SingleInstance force
> #Persistent
> #include Lib\AutoHotInterception.ahk
> 
> AHI := new AutoHotInterception()
> id1 := AHI.GetKeyboardId(0x04F2, 0x0112, 1)
> cm1 := AHI.CreateContextManager(id1)
> 
> id2 := AHI.GetKeyboardId(0x25A, 0x001F, 1) 
> cm2 := AHI.CreateContextManager(id2)
> return
> 
> #if cm1.IsActive
> 
> ; ...
> 
> #if
> 
> #if cm2.IsActive
> 
> ; ...
> 
> #if
> ```

It works!Thank you very very much,you just save my life!
I will make a tutorial on Youtube to show that AHI is awesome.
evilC,You are genius!Thanks.

closed 2019-12-11 by PotatoJet