evilC/AutoHotInterception #53: Help!Context mode have a big BUG!

PotatoJet, 2019-12-11:
Sometimes,it won't blocked.
For example,I change "z" to "O",then I test it in Word.However,sometimes it still show "z".
My script and screenshot ↓↓↓
`#SingleInstance force
#Persistent
#include C:\AutoHotInterception\Lib\AutoHotInterception.ahk

AHI := new AutoHotInterception()

id1 := AHI.GetKeyboardId(0x258A, 0x001F, 1)
cm1 := AHI.CreateContextManager(id1)
return

#if cm1.IsActive

z::O
	
#if`
[image]
Please help me fix the problem，I'm very grateful.

IronManTom, 2019-12-12:
@evilC Well,I also face this problem，I can't figure out how to solve it.

evilC, 2019-12-15:
Context mode can sometimes potentially be flaky, try using Subscription mode and see if issues persist

PotatoJet, 2019-12-16:
> Context mode can sometimes potentially be flaky, try using Subscription mode and see if issues persist

I know Subscription mode is good,but Context mode is much more easy and clean than Subscription mode.
Could you please fix the Context mode problem？Thank you so much.

evilC, 2019-12-16:
I had a look at it and indeed you are correct, Context Mode is simply unreliable.  
I tried to fix it, but I cannot seem to make it at all reliable, I am thinking that I will probably have to remove this feature :(

PotatoJet, 2019-12-17:
> I had a look at it and indeed you are correct, Context Mode is simply unreliable.
> I tried to fix it, but I cannot seem to make it at all reliable, I am thinking that I will probably have to remove this feature :(

All right,I will use subscription mode.
How to change the below context script to subscription script?
(Only subscription,not combined.Thanks)

```
#SingleInstance force
#Persistent
#include C:\AutoHotInterception\Lib\AutoHotInterception.ahk

AHI := new AutoHotInterception()
id1 := AHI.GetKeyboardId(0x1C4F, 0x0002, 1)  
cm1 := AHI.CreateContextManager(id1)

id2 := AHI.GetKeyboardId(0x258A,0x001F, 1)   
cm2 := AHI.CreateContextManager(id2)
return

#if cm1.IsActive

z::
send A
return

#if

#if cm2.IsActive

z::
send O
return

#if

```

PotatoJet, 2020-06-12:
@evilC I am sorry to bother you again.Could you please fix Context mode's problem?Thanks.
I know AHI is good,really good.
I am using Hasu usb to usb converter now,but Taran's script still have big bug,which drive me crazy.

PotatoJet, 2020-06-14:
@evilC  Please help me.No one can help me,only you.
I really apreciate it.

PotatoJet, 2020-06-17:
@evilC Anyhow,take good care of your health.You are a good man.

PotatoJet, 2020-07-13:
@evilC Please help me.

evilC, 2020-07-13:
I tried and failed - it's almost impossible to debug interop between C# and AHK
There's nothing that can be done in context mode that can't also be done in subscription mode, so I suggest you use that

PotatoJet, 2020-07-13:
All righ,it's ok.You are a good man.

closed 2020-07-13 by PotatoJet