evilC/AutoHotInterception #54: Keys not intercepted

jvachez, 2019-12-15:
Hello

I have a XP Pen Artist 15.6 Pro. My tablet is detected under keyboard in AutoHotInterception Monitor as HID\VID_28BD&PID_090D&REV_0000&MI_00&Col03 and also in mices but keys and mouse clicks are not detected even if I check everything.

PotatoJet, 2019-12-15:
Hello,can you help me with the #53 issue?

evilC, 2019-12-15:
I never heard of a tablet device sending keys.  
Maybe the software that comes with the device is sending key presses, in which case AHI would not support them, as AHI only supports input that comes directly from a hardware device

jvachez, 2019-12-16:
Ok thanks
The tablet have buttons associated with keyboard shortcuts in a driver.
I thought I was working because AHKHID detects the keys.

closed 2023-03-21 by evilC