evilC/AutoHotInterception #57: Hello evilC,How to change the below context script to subscription script?

PotatoJet, 2019-12-19:
I'm sorry to bother you again.
How to change the below context script to subscription script?
(Only subscription,not combined.Thanks)

```
#SingleInstance force
#Persistent
#include C:\AutoHotInterception\Lib\AutoHotInterception.ahk

AHI := new AutoHotInterception()
id1 := AHI.GetKeyboardId(0x1C4F, 0x0002, 1)  
cm1 := AHI.CreateContextManager(id1)

id2 := AHI.GetKeyboardId(0x258A,0x001F, 1)   
cm2 := AHI.CreateContextManager(id2)
return

#if cm1.IsActive

z::
send A
return

#if

#if cm2.IsActive

z::
send O
return

#if

```

PotatoJet, 2019-12-24:
@evilC  Merry Christmas！

evilC, 2019-12-24:
```
#SingleInstance force
#include Lib\AutoHotInterception.ahk

AHI := new AutoHotInterception()
id1 := AHI.GetKeyboardId(0x1C4F, 0x0002, 1)  
id2 := AHI.GetKeyboardId(0x258A, 0x001F, 1)   

AHI.SubscribeKey(id1, GetKeySC("z"), true, Func("Kb1Event"))
AHI.SubscribeKey(id2, GetKeySC("z"), true, Func("Kb2Event"))
return

Kb1Event(state){
	if (state){
		Send A
	}
}

Kb2Event(state){
	if (state){
		Send O
	}
}
```

PotatoJet, 2019-12-24:
> ```
> #SingleInstance force
> #include Lib\AutoHotInterception.ahk
>
> AHI := new AutoHotInterception()
> id1 := AHI.GetKeyboardId(0x1C4F, 0x0002, 1)
> id2 := AHI.GetKeyboardId(0x258A, 0x001F, 1) [...]

Thanks again.
You are a real gentleman.

PotatoJet, 2020-07-20:
[image]

Thank you for your help,evilC.
How to replace the Highlights " (state){ if (state){ "  to  " :: " ?
Because I prefer AHK's normal script.