evilC/AutoHotInterception #59: In Subscription Mode, there is no way to execute commands when a combination is pressed

dawidmiara, 2020-02-04:
I'm experiencing issues in Context mode, described in Issue 53 - so I tried to change my code to use Subscription mode.

Unfortunately, I can't figure out how to write commands for combinations like ^!Numpad5 because subscription mode seems to separately execute commands every time a key is pressed.

Can You help me please?
+AHI is really amazing, thank You

cgalway-iflo, 2020-02-04:
AHI Subscription mode cannot interpret AHK modifier symbols such as `^` and `!`, or process multi key combinations.
If you wish to detect such key combinations, you would need to declare a subscription to CTRL, a subscription to ALT and a subscription to Numpad5, and then work out yourself when all 3 keys are held

closed 2020-02-06 by dawidmiara

LakeishaKowalczyk, 2022-12-31:
Same question, are you figured out, buddy?

wirytiox, 2024-11-22:
same here. i cannot figure it out. any ideas?