evilC/AutoHotInterception #60: Monitor.ahk does not run

AkazaRenn, 2020-04-07:
Tried running Monitor.ahk with AHK, it successfully checked the state of AutoHotInterception.dll being block. After I unblock it, no any MsgBox, but an AHK icon in tray disappearing right after I move the mice onto it. No new window or anything, seems it just stopped right after there's an icon.

evilC, 2020-04-09:
So it gets to the [end of the AHI constructor](https://github.com/evilC/AutoHotInterception/blob/master/Lib/AutoHotInterception.ahk#L51) (ie completes `OkCheck()`) OK?

AkazaRenn, 2020-04-09:
@evilC Tried adding MsgBox around `OkCheck()`, can confirm it passed `OkCheck()`

evilC, 2020-04-10:
Where do you have the AHI files? You didn't put them in Program Files or some other protected folder did you? What happens if you run scripts as admin?

AkazaRenn, 2020-04-10:
@evilC Run as Admin -> same result; It is placed in `Downloads\AutoHotInterception`

Hunter310, 2021-04-14:
Problema con el monitor

[image]