evilC/AutoHotInterception #64: 0x80004002 - No such interface supported

Okaghana, 2020-04-11:
Hey there. I'm trying to send a string via AHI, but I keep getting error messages. Here is my existing code: 

```
#Persistent ; (Interception hotkeys do not stop AHK from exiting, so use this)
#include <AutoHotInterception>

global AHI := new AutoHotInterception()
keyboard = AHI.GetKeyboardId(0x1017, 0xA003)

SendText(string){
	For index, char in StrSplit(string)
	{
		AHI.SendKeyEvent(keyboard, GetKeySC("a"), 1)
		AHI.SendKeyEvent(keyboard, GetKeySC("a"), 0)
	}
}

SendText("TEST")
```

As soon as I run it, the script stops as tells me 

> 0x80004002 - No such interface supported

, followed by an excerpt from AutoHotInjection.ahk.

The rest of AHI works fine, the examples work without a problem

~Okaghana

evilC, 2020-04-12:
`keyboard` in your function is a local variable, not global

closed 2020-04-12 by evilC

Okaghana, 2020-04-13:
That doesn't fix the error. I still get 

> 0x80004002 - No such interface supported

Same for passing the keyboard as a function parameter

evilC, 2020-04-14:
`AHI.SendKeyEvent(keyboard, GetKeySC(%char%), 1)` is also wrong
Function calls (`GetKeySC`) are already in "expression syntax", so wrapping `char` in % symbols is performing a double dereference
`AHI.SendKeyEvent(keyboard, GetKeySC(char), 1)`

Okaghana, 2020-04-14:
Also doesn't fix it.

evilC, 2020-04-14:
`keyboard = AHI.GetKeyboardId(0x03EB, 0xFF02)` should be `keyboard := AHI.GetKeyboardId(0x1017, 0xA003)`
You were literally setting the variable `keyboard` to the text `AHI.GetKeyboardId(0x1017, 0xA003)`, not to the value that `GetKeyboardId` returns

Okaghana, 2020-04-14:
That fixes is. Thank you!