evilC/AutoHotInterception #72: Keyboard is reported twice in monitor, same VID/PID, but different handle

Eugeniu-N, 2021-01-18:
In the monitor my keyboard is reported twice with this data:

ID: 1
VID/PID: 0x1B1C, 0x1B0A
Handle: HID\VID_1B1C&PID_1B0A&REV_0103&MI_00

ID: 2
VID/PID: 0x1B1C, 0x1B0A
Handle: HID\VID_1B1C&PID_1B0A&REV_0103&MI_02

When I press any key the monitor only reports data from the second one, so I'm not sure if this is expected or something specifically on my end or a bug

This causes issues when executing `AHI.GetKeyboardId(0x1B1C, 0x1B0A)` because this function returns 1, while the "active" keyboard has ID 2

ChrisAcrobat, 2023-01-07:
I have a similar problem, but mine is for two different keyboards.
>ID: 15 (Corded Microsoft keyboard)
VID/PID: 0x046D, 0xC52B
Handle: HID\VID_046D&PID_C52B&REV_**1203**&MI_01&Col01

and
>ID: 16 (Wireless Logitech K400+)
VID/PID: 0x046D, 0xC52B
Handle: HID\VID_046D&PID_C52B&REV_**2411**&MI_01&Col01

For me this is a huge problem, because I want to know when keyboard ID 16 is used and not.

ChrisAcrobat, 2023-01-07:
After a few minutes of more searching I solved by using `FromHandle` functions. Is this a stable way of dong things?
```
keyboardId := AHI.GetKeyboardIdFromHandle("HID\VID_046D&PID_C52B&REV_2411&MI_00")
mouseId := AHI.GetMouseIdFromHandle("HID\VID_046D&PID_C52B&REV_2411&MI_01&Col01")
```

evilC, 2023-03-21:
> After a few minutes of more searching I solved by using `FromHandle` functions. Is this a stable way of dong things?
> 
> ```
> keyboardId := AHI.GetKeyboardIdFromHandle("HID\VID_046D&PID_C52B&REV_2411&MI_00")
> mouseId := AHI.GetMouseIdFromHandle("HID\VID_046D&PID_C52B&REV_2411&MI_01&Col01")
> ```

Correct. For devices which show twice in the Monitor, but only one works, then use the `FromHandle` method to always pick the one that works

closed 2023-03-21 by evilC