evilC/AutoHotInterception #73: destroy context

serzh82saratov, 2021-01-31:
Hey!
Is it possible to cancel the context?
How can I call interception_destroy_context.

serzh82saratov, 2021-01-31:
If I do this for everyone
try AHI.CreateContextManager (0)
Or so for one
AHI.Instance.SetContextCallback (4, {})
Can this affect stability?

evilC, 2021-01-31:
If it doesn't crash when you do it, I guess not, but TBH I would be kind of surprised if it worked.
Certainly, once you add a ContextManager, just setting the function object for that device to nothing ie `AHI.Instance.SetContextCallback (4, {})` will not stop the code from thinking that there is a context manager for that device, and it will attempt to fire it.
However, a subscription always takes priority over context. If there is a subscription present for a given binding, then it will not attempt to fire the context manager for that input

evilC, 2021-01-31:
There's a specific command for turning AHI on and off - `SetState()`

serzh82saratov, 2021-01-31:
I haven't seen it before, is it in the new version?
I downloaded v0.5.3, tried it AHI.SetState(0) and my script freezes.

evilC, 2023-03-21:
I see no problem with the following code:
```
AHI := new AutoHotInterception()
enabled := true
keyboardId := AHI.GetKeyboardId(0x04F2, 0x0112)
cm1 := AHI.CreateContextManager(keyboardId)
x::
    enabled := !enabled
    AHI.SetState(enabled)
    return

#If cm1.IsActive
1::
	ToolTip("KEY DOWN EVENT @ " A_TickCount)
	return

1 up::
	ToolTip("KEY UP EVENT @ " A_TickCount)
	return
#If
```
Hit 1 on the keyboard - tooltip shows.
Hit x - tooltip stops showing when you hit 1
Hit x again - tooltip starts showing when you hit 1