evilC/AutoHotInterception #77: AHI.SendKeyEvent doesn't seem to set the .isActive context for that device

mmelon, 2021-08-14:
If I send a key as id1, shouldn't it trigger this hotkey
2::
AHI.SendKeyEvent(id1, GetKeySC("1"), 1)
return

#if cm1.isActive
1::
tooltip,it worked
return
#if

or it might be a problem with sendkeyevent not respecting send level?

Basically I get get my context hotkeys to trigger from sendkeyevent

mmelon, 2021-08-14:
actually i apologise it does work. I just can't get it to work across two different scripts. So script A is sendkeyevent ing and script B is #if cm1.isActive

I can't get script A to trigger script B

evilC, 2021-08-14:
Only one script at a time can be interacting with each keyboard, so this would never work
I am surprised even that it does work when does as part of one script - I certainly did not design it to support that use-case
If you want one script to trigger actions in another script, sending keys from one script to be read from another is a poor way to do it anyway - use something like `talk()` https://www.autohotkey.com/boards/viewtopic.php?t=67939

mmelon, 2021-08-14:
thanks for that. Talk looks very useful.

closed 2021-08-14 by mmelon