evilC/AutoHotInterception #80: Unable to find AutoHotInterception.dll

kassimir, 2021-10-04:
I've tried cloning the repo, and I've tried downloading the folder. I've followed the instructions of downloading the Francisco Lopez interceptor and building it to the best of my ability and grabbing the interceptor.dll files from the x64 and the x86 folders and putting them in the respective lib folders. But when I run Monitor.ahk, I get

`Unable to find D:\AHK\AHI\Lib\AutoHotInterception.ahk\..AutoHotInterception.dll, exiting...` And it's not wrong. That doesn't exist. Could you tell me how to solve this please?

evilC, 2022-01-12:
You need to download AHI from the [Releases Page](https://github.com/evilC/AutoHotInterception/releases)

closed 2022-01-12 by evilC

OSsqirat, 2024-02-24:
Fix Unable to find AutoHotInterception.dll

Just download it from an older version on the Release Page and put it in the Lib folder... and it should work... I believe The new version forgot to add that folder.

I formatted my pc and downloaded the newer version to this date and it did the same issue because in the newer version the "AutoHotInterception.dll" in not included in the lib folder for some reason... 

once you add it from a previous version to the newest version ... it should be running good.

[image]