evilC/AutoHotInterception #88: Touchpad laptop not detected

killerpapy, 2022-04-22:
Hello,

I need to use touchpad laptop with AHI but touchpas action are not detected. 

Do you have any solution ? 

Thanks!

evilC, 2022-04-22:
What do you mean by "Touchpad action"?
Does the Monitor app detect the touchpad?

killerpapy, 2022-04-24:
Hello,

The touchpad action that i need is : the simple tap on the touch pad.
Monitor app detect the touchpad and the click on buttons above the touch pad.
But monitor app doesn't detect simple tap, double tap or click on the touchpad.

Picture of the touchpad : 
[image]

evilC, 2022-04-25:
It's most likely that those functions are handled at a software level by the OS, not at a hardware level.
Interception can only speak directly to the hardware, so that's why it's not seeing those events

killerpapy, 2022-05-02:
Ok thanks for the explanation !