evilC/AutoHotInterception #89: subscribe all swallows it's own synthesised output

mmelon, 2022-05-27:
Hi,

Just checking this is right. When i use

AHI.SubscribeKeyboard(id1, true, Func("KeyEvent"),false)

I can't

AHI.SendKeyEvent(id1, code, state)

because the subscription swallows the synthesised output? Or is there another reason I can't generate a keypress from within the subscribe keyEvent function?

Basically I want to subscribe all per device, then set a variable before allowing the key through. If I turn blocking off, then key is allowed through before the variable has been set.

I hope I have explained this ok.

thx
Mike

evilC, 2022-05-27:
No, AHI will never swallow it's own synthesized output. You're doing something wrong.
In `KeyEvent()`...
Is `AHI` in scope?
Is `id` in scope?

mmelon, 2022-05-27:
thx. AHI wasn't in scope. Feel a bit stupid but your response saved me so much time. thx again.