evilC/AutoHotInterception #95: Disable touchpad laptop if at least one external mouse is existing

hsayed21, 2022-11-17:
My script for testing on `Notepad`

```ahk
#SingleInstance force
#Persistent
#include ..\Lib\AutoHotInterception.ahk

AHI := new AutoHotInterception()
mousedId := AHI.GetMouseIdFromHandle("ACPI\VEN_SYN&DEV_2B96")

SetTimer, WatchWin, -0
return

; 0: Left Mouse
; 1: Right Mouse
; 2: Middle Mouse
; 3: Side Button 1
; 4: Side Button 2
; 5: Mouse Wheel (Vertical)
; 6: Mouse Wheel (Horizontal)

MouseButtonEvent(code, state){
	ToolTip % "Mouse Button - Code: " code ", State: " state	
}

DoSub(state){
	global AHI, mousedId
	if (state){
		AHI.SubscribeMouseButtons(mouseId, true, Func("MouseButtonEvent"))
	} else {
		AHI.UnsubscribeMouseButtons(mouseId)
	}
}

WatchWin:
	Loop {
		WinWaitActive, ahk_class Notepad
		DoSub(true)
		WinWaitNotActive, ahk_class Notepad
		DoSub(false)
	}
	return

^Esc::
	ExitApp
```

**Error:**
```
Error in #include file "d:\AutoHotInterception\Lib\AutoHotInterception.ahk":
     0x80004002 - No such interface supported

Specifically: SubscribeMouseButtons

	Line#
	184: {
	185: this.Instance.SubscribeMouseButton(id, btn, block, callback, concurrent)  
	186: }
	188: {
	189: this.Instance.UnsubscribeMouseButton(id, btn)  
	190: }
	192: {
--->	193: this.Instance.SubscribeMouseButtons(id, block, callback, concurrent)  
	194: }
	196: {
	197: this.Instance.UnsubscribeMouseButtons(id)  
	198: }
	200: {
	201: this.Instance.SubscribeMouseMove(id, block, callback, concurrent)  
	202: }

Continue running the script?Error in #include file "d:\AutoHotInterception\Lib\AutoHotInterception.ahk":
     0x80004002 - No such interface supported

Specifically: UnsubscribeMouseButtons

	Line#
	188: {
	189: this.Instance.UnsubscribeMouseButton(id, btn)  
	190: }
	192: {
	193: this.Instance.SubscribeMouseButtons(id, block, callback, concurrent)  
	194: }
	196: {
--->	197: this.Instance.UnsubscribeMouseButtons(id)  
	198: }
	200: {
	201: this.Instance.SubscribeMouseMove(id, block, callback, concurrent)  
	202: }
	204: {
	205: this.Instance.UnsubscribeMouseMove(id)  
	206: }

Continue running the script?
```
@evilC

evilC, 2023-03-21:
Double-check that you are getting a value in `mouseId`
The "No such interface supported" error usually means you have an empty parameter

hsayed21, 2023-03-28:
I simplified it to be Unsubscribe only but when clicking on buttons touchpad it clicks normal not disabled
```ahk
#SingleInstance force
#Persistent
#include Lib\AutoHotInterception.ahk

AHI := new AutoHotInterception()
mouseId := AHI.GetMouseId(0x0000, 0x0000)
AHI.UnsubscribeMouseButtons(mouseId)
return

^Esc::
	ExitApp
```
[image]

hsayed21, 2023-03-28:
This is after the change with `handle`, touchpad click also work
[image]