evilC/AutoHotInterception #98: Dramatic time increase for Subscribing to keys on newer version

Anutim, 2022-12-23:
Hey, I just switched on to moving my AutoHotInterception scripts to v2, so figured I'd get the latest version too.

I repeatedly Subscribe and Unsubscribe depending on if a condition is met, and upon finishing transition to the latest version, I realized that Subscribing multiple keys in a row is taking ages.

At first I figured I had probably ported AHI to v2 incorrectly somehow, so I went back to v1 and tried the latest version there.
Exact same issue.

To further detail what happens:
Once subscriptions have been done (logging the functions in AHK, it seemingly goes very fast), pressing any of the subscribed keys won't do anything for a while (as extreme as 10 subscriptions taking like 10 seconds?).
Until suddenly, all the keystrokes appear to be processed at the same time, creating a huge spam.
Afterwards it works as normal, until it's time to re-subscribe.

Old version I was on (0.5.0), there wasn't really any noticeable lag after subscribing, it felt instant.
And I haven't tried any version in-between until latest (0.8.0), so couldn't really say when the problem was introduced.

LAPIII, 2023-02-12:
Can you upload  your ported AHI to v2?

Anutim, 2023-02-12:
https://cdn.discordapp.com/attachments/195652955662319636/1074391409425141830/AutoHotInterception.ahk (0.5.2)
https://cdn.discordapp.com/attachments/195652955662319636/1074391423153098762/CLR.ahk
https://cdn.discordapp.com/attachments/195652955662319636/1074391712820105288/anuAHI.ah2

Holding down a key will keep sending multiple calls similar to Windows default "Hold key to spam it" functionality, so this is what I do to avoid that (per function):
```
Function(state){
        static previousState := 0
        if (state == previousState)
            return
        previousState := state

        if (state)
            ThingToDo()
}
```

evilC, 2023-03-21:
I have an initial branch of AHI for AHK v2 [here](https://github.com/evilC/AutoHotInterception/tree/ahk-v2-support)
If you can reproduce issues mentioned with this branch, please let me know

Anutim, 2023-03-21:
Issue persists.

Also, unsure why, but switching to it also randomly crashes two games I've tried while using it?
No idea what's going on there or how to start troubleshooting it.

MartinLichtblau, 2023-08-30:
I have the same issue. Latest AHI version [v0.9.1](https://github.com/evilC/AutoHotInterception/releases/tag/v0.9.1) with AHKv1.

MartinLichtblau, 2023-08-31:
[v0.7.0 ](https://github.com/evilC/AutoHotInterception/releases/tag/v0.7.0) works fine. So bug was introduced in [v0.8.0 ](https://github.com/evilC/AutoHotInterception/releases/tag/v0.8.0).?

Spiceries, 2026-01-13:
With the latest version, I am encountering the same problem. Subscribing keys is very very slow.