evilC/AutoHotInterception #102: FileInstall fails if the library files already exist

tpensyl, 2023-05-30:
When I compile my script to an executable, it gives the below error.
```
Call stack:
*#1 (109) : [FileInstall] FileInstall("Lib\AutoHotInterception.dll", "Lib\AutoHotInterception.dll")
*#1 (109) : [AutoHotInterception.Prototype.__New] FileInstall("Lib\AutoHotInterception.dll", "Lib\AutoHotInterception.dll")
*#1 (317) : [Object.Call] AHI := AutoHotInterception()
*#1 (317) : [] AHI := AutoHotInterception()
> Auto-execute
```
After experimenting it seems to be a problem trying to create the files that already exist within Lib.  And even if I run the executable in  a  fresh folder  it only works the first time, failing on subsequent attempts since libalready exists

I don't know if I'm doing something wrong, but  I was able to resolve the issue for myself by modifying 
AutoHotInterception.ahk to set the overload flag to 1 in the 3 invocations to FileInstall().  I'm using v2. If this is a general issue,  considering adding the overload flag by default to the script.

Luciener, 2023-06-17:
Hello. I have a similar error, only under the number 111 (V2). Can you post the modified code for the file "AutoHotInterception.ahk"?

tpensyl, 2023-06-17:
the issue is now fixed in master:
https://github.com/evilC/AutoHotInterception/commit/c218f9f2b87f5e30bda24682d3a9debdadf25209

tpensyl, 2023-06-17:
Oh I see now, this was already fixed in master weeks before I hit the issue, but it is not yet included in the latest release (0.9.0)

closed 2023-06-17 by tpensyl

Luciener, 2023-06-18:
Thanks for information.