evilC/AutoHotInterception #109: Meida keys not properly intercepted (AHK sees the, but AHI script does not)

Av4sX, 2024-01-28:
AHI can not intercept media keys such as volume up, stop, sleep, home etc.
Used monitor script - it shows no activity when pressing those buttons. Nor ot in anyway reacts when using exaple subscription all keys script. Whn I open AHK history - I can see those buttons, but somehow script does not react on them.... 
Need help because all the idea of AHK and AHI was to change media keys (especially sleep and airplane mode) to my combinations and so not let system react on theme (aka subscribe keys). 

I tried to use sample subscribe one key script - it works with "1" or similar keys, but when I put "Volume_Down" there it does no scubscribe/work with it.

Av4sX, 2024-01-28:
Important nuance: AHI script does NOT react on bedia keys IN BOTH BODES - context and subscribe. I can understand if it doesnt work with USB keyboards, but as instruction said it only qualifies for subscibe mode. Why then context mode does not see media keys?

Av4sX, 2024-01-28:
Also info: When tested on another device (notebook) it successfully got Vlume Up/Down BUT still did not react on Sleep and Airplane modes. As well as special "rog" keys. AHI sees nothing there.

evilC, 2024-02-24:
The interception driver which AHI relies upon does not support the media keys, so unfortunately, AHI is unable to support them as a result.  
Actually, I think the limitation is that it does not support them for USB devices - it may support them via PS/2.

closed 2024-02-24 by evilC