evilC/AutoHotInterception #113: cm1.IsActive returns "0"

KenKaneki73985, 2024-05-03:
Everything is set, except "cm1.IsActive", it returns "0"
AutoHotInterception v0.9.1

`; #SingleInstance force
Persistent
#include Lib\AutoHotInterception.ahk

AHI := AutoHotInterception()
id1 := AHI.GetKeyboardId(0x09DA, 0x0025, 1)  
cm1 := AHI.CreateContextManager(id1)

isset(AHI) ? msgbox( "AHI is set") : msgbox("AHI not set")
isset(id1) ? msgbox("id1 is set") : msgbox("id1 not set")
isset(cm1) ? msgbox("cm1 is set") : msgbox("cm1 not set")

msgbox(cm1.IsActive)   ; returns 0

return
#HotIf cm1.IsActive

::aaa::JACKPOT
1::
{
	ToolTip("KEY DOWN EVENT @ " A_TickCount)
	return
}

1 up::
{
	ToolTip("KEY UP EVENT @ " A_TickCount)
	return
}
#HotIf

^Esc::
{
	ExitApp
}
`

evilC, 2024-05-03:
Because `IsActive` is only ever true for a VERY brief moment. 
It is not intended to indicate whether the context manager is actively running.

AHI just sets `.IsActive`  to true before sending a keystroke from that specific keyboard, so that the context sensitive hotkey in the script executes for only that keyboard.
ie When keyboard input happens for a keyboard with an active Context Manager, AHI sets `.IsActive` to true, then forwards on that key event to the OS, then immediately sets `.IsActive` back to false again.

KenKaneki73985, 2024-05-03:
@evilC  Okay, thanks.

So, everyting seems working fine on my PC.

The "monitor.ahk" is working fine. I was able to detect the VID / PID of my keyboard.

However, pressing "ENTER" doesn't execute the action below. Instead, It does its native action rather than sending "hello".

What do you think is the problem?

  ; #SingleInstance force
  Persistent
  #include Lib\AutoHotInterception.ahk
  
  AHI := AutoHotInterception()
  id1 := AHI.GetKeyboardId(0x09DA, 0x0025, 1)  ; 0x09DA, 0x0025 <------ ID of my keyboard
  cm1 := AHI.CreateContextManager(id1) 
  
  return
  
  #HotIf cm1.IsActive
  
  ENTER:: {
      send("hello")
      return
  }
  #HotIf

evilC, 2024-05-03:
When in the monitor, you find your keyboard, and check some keys, it would look like this:
[image]
(eg you can see me hitting CTRL+Printscreen in that screenshot)
The ID that is selected is 3
In your code, check what the value of `id1` is. In the example screenshot, I would expect it to be `3`

evilC, 2024-05-03:
Also check the instance number. Note how in my screenshot above, both ID 2 and 3 have the VID/PID of 0x03EB, 0xFF02
However, only if I check ID 3 do I see input in the monitor.
So `id1 := AHI.GetKeyboardId(0x03EB, 0xFF02, 1)` will not work for me, I would either have to use `id1 := AHI.GetKeyboardId(0x03EB, 0xFF02, 2)` (Note the 2), or use the Handle instead of VID/PID (`id1 := AHI.GetDeviceIdFromHandle("HID\VID_03EB&PID_FF02&REV_0090&MI_03")`

KenKaneki73985, 2024-05-03:
@evilC Hi. It's working now. Thanks!

However, I'm confused about the third argument "2":
AHI.GetKeyboardId(0x09DA, 0x0025, 2) ; <---- WORKING - my actual keyboard ID

Monitor shows that my keyboard ID is 4, but I passed "2" instead "4"?

[image]

evilC, 2024-05-03:
See in your monitor: Both ID 2 and ID 4 have the same VID / PID: `0x09DA, 0x0025`
So to use ID 4, you need to say "Don't use the 1st 0x09DA, 0x0025, use the 2nd 0x09DA, 0x0025"
Which is what the `2` at the end is - it says "The 2nd one".
ie either `AHI.GetKeyboardId(0x09DA, 0x0025)`  or `AHI.GetKeyboardId(0x09DA, 0x0025, 1)`  will both return `2`  
However, `AHI.GetKeyboardId(0x09DA, 0x0025, 2)`  will return `4`

evilC, 2024-05-03:
It's also interesting to note that the handles for 2 and 4 are not unique, which is not something I have ever seen.
eg notice in my screenshot - whilst the VID and PID for ID 2 and 3 are the same (0x03EB, 0xFF02), their handles are different (HID\VID_03EB&PID_FF02&REV_0090&MI_01 and HID\VID_03EB&PID_FF02&REV_0090&MI_03).
So normally, an alternative to passing a 2 at the end of `GetKeyboardId` would be to use `GetDeviceIdFromHandle` instead, but seeing as you have two devices with the same handle (Or, more likely, one device that presents multiple IDs / Handles), that would not work.

evilC, 2024-05-03:
Closing as completed

closed 2024-05-03 by KenKaneki73985